waydabber/m1ddc #3: Mac mini m1 - HMDI port

AM-PRO, 2021-08-30:
Hello,

Thanks for your work, it helps me a lot !

I'm trying to make this package works with the mac mini m1 hdmi port, do you plan to work on it ? If not, can you please give me some advices/tracks that i can work on ?

THANKS !

waydabber, 2021-08-30:
Hi! Are you a technical guy who is trying to figure out the problem or an end user who is waiting for a solution?

If you are an end user, I recommend you buying a USB-C to HDMI adapter. Most of these work (some don't but that's the exception and in this case you can ask for a replacement to a different one). Unfortunately the mini's HDMI port seems to blocks DDC communication for some reason. This is true for the 2018 Intel mini as well. You can only use software dimming on displays connected via HDMI on the mini.

If you are a technical guy and want to dig deeper to the issue, then the issue is that i2c communication (which is used by DDC) with chip address 0x37, data address 0x51 does not seem to work through the HDMI port of the M1 mini. Some kind of communication must be going on since the display can read EDID data from the display (this uses address 0x50) and during testing there were instances when HDMI displays behaved strangely when attempting DDC communication through the M1 HDMI port (I experienced this myself - it made my LG display go crazy and I needed to do a power cycle to fix it). Also some guy said that with a 2018 mini he was able to issue some DDC commands briefly after connecting the display, before the communication was blocked, but I can't find the source of this info, so it might be a heresay. 

There might be a way to solve the problem but we didn't succeed so far (but since what we have is also a result of blind attempts, reverse engineering and some luck, this means nothing - the solution might be just around the corner, we never know until somebody stumbles on it). Also, we didn't try too hard yet - we had plan with @alin23 to iterate through all possible addresses and try some random things to figure out what works and what doesn't but this didn't go too far yet (mainly because of me, the process is super tedious).

AM-PRO, 2021-08-30:
Thanks for your reactivity and for all these informations that you just gave me.

I'm trying to setup a PC/Mac mini desktop with 2 monitors / 1 mouse / 1 keyboard. I have a USB switch for my mouse and keyboard and i'm trying to figure out how to make a little program that can help me switch monitors inputs.

Everything is working fine on Windows and i can switch to Mac os. 
If I try to switch from mac to windows, I can use m1ddc to switch the display port, but it doesn't work with the HDMI port.

I'm a tech guy but I don't have much knowledge in Apple ecosystem. If I can help i'll do it.

I'll try with a USBc to HDMI cable and come back soon.

waydabber, 2021-08-30:
Ok, good luck then! :)

closed 2021-12-10 by waydabber