waydabber/m1ddc #4: Request control

abizeauMaxon, 2021-09-27:
Your program work perfectly, but Im facing a little issue with a special setup.

**Error:**

`I2C communication failed: (iokit/?) unknown subsystem error`

**Probblem description**

I have a monitor plug with input 18 on a Windows machine, and input 17 on a macbook pro m1 2020.

While being on the mac, if I do m1ddc display 1 set input 18, everything is fine, and the control of the screen is sent to the Windows machine. But if I try on the mac do request the control of the monitor using m1ddc display 1 set input 17, while the windows machine having it.

Note that I have a similar program as m1ddc on windows, and it can request the control of the screen even if the mac has it, so it seems to be feasible, but might also be an apple restriction, i dont know.

waydabber, 2021-09-27:
Hi, it depends on whether the display reports itself available to the mac or not. If it does not, then macOS removes the display from the ioreg tree and you can't send commands to it any longer. Or if it does not the display might still not receive I2C commands from the display. Some displays behave differently regarding this on their HDMI, DisplayPort and USB-C ports. Also, macOS might be more finicky regarding this than Windows.

abizeauMaxon, 2021-09-27:
If I do the m1ddc display list, it being listed even while Windows has the control. But 'input' command not working has the error shown above.

waydabber, 2021-09-27:
If this is the case, then the display was not removed from the ioreg tree but I2C communication (which is a simple serial communication protocol used by DDC) is blocked either by the display or by the mac. You might want to try switching which machine uses which input on the display and see if mac can control the display as if it were in the place of the Windows machine while the display is set to Windows.

closed 2021-12-10 by waydabber