waydabber/m1ddc #8: Failing to make on 12.1

pqhf5kd, 2021-12-19:
I'm getting the following error on 12.1
`clang -fmodules -o m1ddc m1ddc.m
m1ddc.m:84:63: error: use of undeclared identifier 'kIOMainPortDefault'
            io_registry_entry_t root = IORegistryGetRootEntry(kIOMainPortDefault);
                                                              ^
1 error generated.
make: *** [all] Error 1`

I see you switched from kIOMasterPortDefault to k10MainPortDefault in #7

waydabber, 2021-12-19:
Hi, it might be that XCode or XCode Tools are not the latest in your system. `kIOMasterPortDefault` was giving a deprecated warnings on the latest versions, that's why I had to change it.

pqhf5kd, 2021-12-19:
`softwareupdate --list` tells me that no new software is available.
`clang --version` returns:
`Apple clang version 12.0.5 (clang-1205.0.22.9)`

I don't have Xcode

closed 2021-12-19 by pqhf5kd

pqhf5kd, 2021-12-19:
Manually installing CLI Tools 13.2 fixed this issue, sorry!