waydabber/m1ddc #9: Power Command

pqhf5kd, 2021-12-19:
Will you be implanting the power/standby command?

ddccdl isn't working on M1 Macs

waydabber, 2021-12-19:
Hi, there is a `standby` command, but it is not listed in the help.

I am planning to replace the m1ddc thing with a new swift version that uses [this library](https://github.com/waydabber/AppleSiliconDDC) I developed for [MonitorControl](https://github.com/MonitorControl/MonitorControl) and supports [all commands](https://github.com/MonitorControl/MonitorControl/blob/main/MonitorControl/Enums/Command.swift) eventually, I just didn't have the time for it so far (or maybe I'll do it as a CLI for MonitorControl).

pqhf5kd, 2021-12-19:
> Hi, there is a standby command, but it is not listed in the help.

Found it, thanks!
Unfortunately, while 1 will wake my display and 2, 3 and 4 will put it to standby, it only works with the 'soft standby' that appears to power off the panel and backlight but leaves the LED flashing.

There aren't separate power and standby commands correct?

waydabber, 2021-12-20:
Hi, no, there is only the powerMode command. But a display might have manufacturer specific additional power control commands (for example powerControl 0xE1).

pqhf5kd, 2021-12-20:
Can I just change line 16 from 0xD6 to 0xE1 before running Make to test this?

I have a Samsung display and a quick google suggestions that 0xE1 is a Samsung command

waydabber, 2021-12-21:
Yes, that should do the trick.

pqhf5kd, 2021-12-21:
I tried but no luck, thanks @waydabber

closed 2021-12-21 by pqhf5kd