waydabber/m1ddc #13: Not able to set input

vwumumu, 2022-06-25:
I am using M1 MAC Air to connect monitor LG 27UN880 with DP USB-C.
I am able to use most of commands such as to get/chg volume, contrast but failed to set input.
The monitor will flash and back to current view after I execute set input 17.
Switch input is working if I use the physical button on the monitor.
Could you please advise if you have any insight into my problem?

waydabber, 2022-06-25:
Hi there,

well, m1ddc certainly sends the proper command - what the display does with it is entirely up to the display firmware. Not all displays support input changing properly, but LG displays generally (not always) do. It might be that number 17 does not correspond to the desired input?

vwumumu, 2022-06-25:
@waydabber  Thanks a lot for your quick response. I found I can use ControlMyMonitor on Windows to change the property of Volume and Contrast like what I can do with m1ddc on MAC Air M1. But failed to change the Input value even though the permission for this property in ControlMyMonitor is Read+Write which is like volume. The possible values of Input Select in ControlMyMonitor is showing 0,17,18,15,0. I have tested these values and got the same problem. The display will flash and back to current view. I am going to search if any new firmware for my display and post updates here.

waydabber, 2022-06-25:
Great, thank you! :)

waydabber, 2022-06-25:
(btw, I'd try input numbers from 0-32 regardless of the reported supported values, as those are not always reliable).

Githubcatone, 2022-11-19:
> @waydabber Thanks a lot for your quick response. I found I can use ControlMyMonitor on Windows to change the property of Volume and Contrast like what I can do with m1ddc on MAC Air M1. But failed [...]

I also met the same problem, have you solved it?

Hacksore, 2022-11-19:
Same issue here. 

I've enumerated 0-32 and it won't switch to the other input. On some values I get the same flash to black @vwumumu was getting then back to current input.

Monitor: LG 27GL850-B connected via DP
macOS: 13.0.1 (22A400)

Going to also look into a firmware update for my monitor.

artus70, 2024-05-12:
> The possible values of Input Select in ControlMyMonitor is showing 0,17,18,15,0. I have tested these values and got the same problem. 

For me, it didn't show the real number options. I've got a Samsung Monitor where these numbers are assigned to the ports:

- HDMI1: 5
- HDMI2: 6
- DisplayPort: 15

I use `winddcutil.exe setvcp 1 96 5` to switch input to HTML1. (Download: https://github.com/scottaxcell/winddcutil.) Note: 96 is the decimal representation of the vcp feature number 60 hexadecimal.

waydabber, 2026-10-29:
LG alternate input switching was added in #32, with the `UN880` among the tested displays. The related write checksum issue was fixed in #52. Both are included in [v1.3.1](https://github.com/waydabber/m1ddc/releases/tag/v1.3.1).

Use `set input-alt 144` for `HDMI 1`, `145` for `HDMI 2`, or `208` for `DisplayPort 1`.

Closing as addressed for the original LG report. Please let me know if the current version still fails!

closed 2026-10-29 by waydabber