waydabber/m1ddc #14: I2C communication failure

commiepinko, 2022-07-02:
I'm running macOS 12.4 on an M1 Mini with a Dell P3222QE.  I get 

`I2C communication failure: (iokit/?) unknown subsystem error`

in response to every m1ddc command.

This is a bit beyond my pay grade - should I assume the problem is with the display?

pgoca, 2022-07-04:
I have the same problem on my M1 macbook pro with AOC Q27G2G4. I also run it on macOS 12.4. The error itself is way beyond my understanding, so I'm just leaving a note that it's not only your display that fails :)

waydabber, 2022-07-04:
Yeah, some displays just won't support DDC. It is either the HDMI port of the Mac (which has issues with DDC) or the display. I have a Lenovo Creator Extreme, it does not support DDC over the USB-C connection (even though it uses DisplayPort alt-mode) but supports it over DisplayPort for example. It is totally display firmware specific.

pgoca, 2022-07-05:
Thanks for the quick response, I will try with the display port

x-magic, 2022-08-07:
I have a Dell U4320Q and it works fine via USB-C DP Alt port, but won't work via DP on a Thunderbolt dock. I guess it has something to do with whether the middle layers (docks, USB hub controllers) support DDC/CI or not.

waydabber, 2022-11-12:
yes, it depends on how the dock handles i2c communication for ddc. Some docks/dongles are just not handling it right or doing it only for EDID data (which is required so the display works at all).

emircanerkul, 2022-11-15:
> I2C communication failure: (iokit/?) unknown subsystem error

@waydabber When I use https://github.com/waydabber/BetterDisplay#readme It works perfectly but when I use this cli, it doesn't. Says same error in my mac m1. I just need a basic HDMI to DP switch command.

waydabber, 2022-11-15:
Hi, can you try the latest BetterDisplay version (that supports ddc input switch - does that work)?

waydabber, 2022-11-15:
Resolution, refresh rate, colors (if you are talking about adjustments) and brightnes (by default for the 0-50% part) does not require DDC. Question is if does DDC brightness (normally 50-100% which controls the hardware backlight), DDC contrast, DDC color control, DDC display input switching etc. work with BD. If not, then the problem is the same as with m1ddc.

emircanerkul, 2022-11-15:
@waydabber yea, you're right. Now I checked the settings under the DDC video adjustment tab, all doesn't work.

waydabber, 2022-11-15:
What connection are you using?

emircanerkul, 2022-11-15:
@waydabber Display Port (LS28AG700N)

waydabber, 2022-11-15:
Well, that should work. :) It might be that the display has some issues or does not support DDC on all ports. I have no experience with the G7.

sidevesh, 2023-07-17:
Seeing this error to and in my case this happens with all monitors I have tried and I can verify that the same monitors through the same dongle / dock works on my Linux or Windows laptop.
Is there a way we can get more info into the error atleast, the current error is really vague.

sidevesh, 2023-07-18:
I am also seeing the error on an Intel MacBook, again, another Windows laptop on the same exact config works normally.

sidevesh, 2023-09-11:
Maybe unrelated but in my Intel MacBook, the MonitorControl app started working after the recent update

duncanfinney, 2024-07-09:
Just a note for other people facing this issue: Seems to work great with Caldigit TB3 dock, but my Anker 777 doesn't pass through the i2c sadly.

waydabber, 2026-10-29:
This is a general `IOKit` error saying the `I2C` transaction failed. It does not identify the cause or necessarily mean the display is faulty. Common causes include disabled `DDC/CI`, a dock or adapter not passing the commands through, or the display not responding on that port or input.

Please use the current version, enable `DDC/CI` in the monitor menu if available, and try a direct connection. M1 built-in `HDMI` support was also added in #60.

Closing this as a hardware compatibility/support question. If a specific failure can be reproduced with the current version and a known working `DDC` connection, a new report with those details is welcome. Thanks!

closed 2026-10-29 by waydabber