waydabber/m1ddc #18: Can DisplayLink connected monitors work?

ubbooo, 2022-11-12:
Not really an issue. More of a question. Is there any hope for monitors connected via DisplayLink drivers to an M2 Air?

I have two external monitors and m1ddc works great from my work M1 Pro MBP (directly connected).

But on my private M2 Air sitting beside with DisplayLink and the two monitors connected via a Dell DS6000 hub then m1ddc does not seem to see any monitors at all. I assume it is the DisplayLink which seems to be a really weird beast?

waydabber, 2022-11-12:
I am afraid it won't work. DisplayLink might implement a custom API for DDC control but currently this feature is not available on the mac. DisplayLink screens are seen as virtual screens (like a Sidecar display or a BetterDisplay dummy display) in macOS.

waydabber, 2026-10-29:
Closing this as answered: `DisplayLink` connections are not supported by `m1ddc`. It uses the Mac's native display `DDC` path, which is not available through these virtual displays. A direct `USB-C`/`DisplayPort` or supported built-in `HDMI` connection is needed. Thanks!

closed 2026-10-29 by waydabber