waydabber/m1ddc #19: get luminance doesn't always return the correct number

FischLu, 2022-11-20:
Hi, I am using two display for MacStudio, one is LG 27UP850, another one is LG 27UP650. I am trying to combine Hammerspoon and m1ddc together. I want to know the current brightness before I change it, however, the command
```
m1ddc display 1 get luminance
```
doesn't always return the current brightness (which is 50 during my test), sometimes it returns 0 and I see one time it returns a negative number, as you can see from the image

[image]

For `get volume` I have the same problem too..

waydabber, 2022-11-20:
well, you should make an iteration in your luascript to get the value 5 times and use the max you get. DDC reading is inherently finicky.

FischLu, 2022-11-20:
thanks, at least I have a solution now.

closed 2022-11-20 by FischLu