waydabber/m1ddc #20: [Feature request] Ability to turn on/off a display

pavelrad, 2022-11-22:
It is possible in Lunar to send a "turn off" signal to a display. Would be great to have the same one on m1ddc, together with "turn on", which I hope should work at least on my Dell U2515H, as it correctly reports its settings via DDC when being turned off.

waydabber, 2022-11-22:
Are you sure it is possible in Lunar? Power control via DDC is possible but rarely implemented correctly and most of the time you can't turn the display back afterwards.

If you mean Lunar's blackout feature, it is not actually turning the display off, just hides it behind a mirror and turns brightness down as much as possible.

pavelrad, 2022-11-22:
I mean turning off using DDC, which works just as expected on my monitor: it turns off like the power button was pressed.

pavelrad, 2022-11-22:
I saw a comment in Lunar on this feature, which says that it might not work, and it's impossible to turn on after that. But I still believe that turning on might work, because as i stated in the first comment, my monitor responds to DDC commands when it turned off. Looks like it is actually in some standby mode. When I disconnect it from the power source, m1ddc is not able to get any response from it and prints some corresponding error.

waydabber, 2022-11-23:
but... power control is highly manufacturer specific so I am not sure if it is a good idea to make it generally available. :)

pavelrad, 2022-11-23:
Wow! It is already here but just not reflected in the help. `standby` worked for me as expected. I can turn on the display with `m1ddc set s 1` and turn it off with `m1ddc set s 5` commands. Thank you for your help! :bow:

closed 2022-11-23 by pavelrad