waydabber/m1ddc #21: Add this to homebrew

anonymousaga, 2022-11-27:
Homebrew is the package manager of the mac. It would be easier to install & update if this came via homebrew. Thanks in advance.

waydabber, 2022-12-18:
That's a great idea! I'll be doing this, but meanwhile if somebody else wants to add the app, here's how to proceed: 

https://docs.brew.sh/Adding-Software-to-Homebrew

anonymousaga, 2022-12-18:
I can help you once i get back from vacation later this week. Thanks!

rogerfantw, 2023-01-06:
@anonymousaga 
Do you have time to add the app to homebrew recently? I'm looking forward to installing this app via homebrew. Thanks a lot.

anonymousaga, 2023-01-06:
Ill try after jan11 … im quite busy till then

unlobito, 2023-01-08:
I've raised https://github.com/Homebrew/homebrew-core/pull/120078 to request m1ddc's addition to Homebrew, but I fear it may be turned down under the [stable versions](https://docs.brew.sh/Acceptable-Formulae#stable-versions) policy since this repo doesn't have tagged releases.

@waydabber Would you be able to create a tagged release for m1ddc? (e.g., `1.0` - [About Releases](https://docs.github.com/en/repositories/releasing-projects-on-github/about-releases))

waydabber, 2023-01-09:
@unlobito - sure, here it is!

https://github.com/waydabber/m1ddc/releases/tag/v1.0.0

unlobito, 2023-01-26:
🚀 https://formulae.brew.sh/formula/m1ddc

waydabber, 2023-01-26:
Nice! :)

anonymousaga, 2023-01-26:
Im not sure what this dies cause im not getting issues, but this error came:

Warning: The post-install step did not complete successfully
You can try again using:
  brew postinstall m1ddc

When i ran that command the same error came. Thanks in advance

closed 2023-01-26 by anonymousaga