waydabber/m1ddc #22: Issues Setting/Getting Values with macOS 13

SomeInterestingUserName, 2022-11-27:
I've been using m1ddc for some time now to automate switching inputs on my monitor, but I discovered after upgrading to macOS 13 that it does not perform as expected. I'm running m1ddc on an M2 MacBook Air, previously on macOS 12, and now on macOS 13.0.1. My monitor is connected via DisplayPort over a USB-C to DisplayPort adapter, which worked with macOS 12. 

On macOS 12, the following command resulted in an input change (switching to the HDMI input on my monitor): 
`m1ddc display 1 set input 5`

However, the same command now fails silently on macOS 13. I made sure to rebuild m1ddc after the OS upgrade. 

Running `m1ddc display list` still shows the display as it has before. 

I also noticed that getting values using `m1ddc display 1 get [x]` (where `[x]` can be `input`, `luminance`, etc) returns erratic values that fluctuate when called repeatedly. For example, running `m1ddc display 1 get input` repeatedly returns 0 most of the time, but sometimes -1, -124, or 16, even if no parameters are modified (either through m1ddc or externally via the monitor's controls). Unfortunately, I didn't get a chance to experiment extensively with `get` on macOS 12 to have a useful baseline for comparison, but intuitively, the fluctuating values aren't consistent with expected behavior.

SomeInterestingUserName, 2022-11-27:
Closing this issue because of a rather trivial fix...unplugging the monitor for a few seconds and plugging it back in 😆   Apparently the monitor must've gotten into a bad state and powering it off (multiple times) with the power button didn't suffice.

closed 2022-11-27 by SomeInterestingUserName

waydabber, 2022-11-27:
lol,  huh, glad it worked out as I'd have had no idea how to fix the problem...