waydabber/m1ddc #26: Controlling monitor when not on same input

mikehy70, 2023-01-22:
I have 2 monitors - a Dell and a TCL.  Changing inputs, adjust volume/brightness all work fine as long as Mac is being viewed on Dell (connected via USBC).  If I switch the dell to another input (through CLI or otherwise) I can no longer have the Dell change  inputs to get back to Mac input.  None of the controls work.  The display is still enumerated as display 2, it's still connected via USBC but can no long control it when you switch inputs.

waydabber, 2023-03-31:
This is up to the display whether it accepts DDC commands from a non-active input. Some do, more don't.

waydabber, 2026-10-29:
Whether an inactive input accepts `DDC` commands is up to the display firmware, so there is no universal fix for that part.

There is a software improvement we can track here though: discovering available `DDC` services even when the display is no longer listed as online by macOS. #61 proposes this and is still open.

Keeping this as an `enhancement` for that discovery support. It cannot make a display respond if its inactive input stops accepting `DDC`, including when the display remains listed as in your original report.