waydabber/m1ddc #37: get input randomly returns 110

sylveon, 2023-09-30:
The get input method doesn't appear particularly reliable: I randomly see `110` when calling the method on two different monitors from two different vendors.

[image]

artus70, 2024-05-12:
+1

`./m1ddc display list detailed` works fine.

`./m1ddc display 1 get c` should display the contrast of display 1 but always shows "110", same on other parameters.

WeaverDev, 2024-11-13:
+1
Working around it by retrying until value is <= 100

waydabber, 2026-10-29:
A related read-request checksum bug was fixed in #62 and is included in [v1.3.1](https://github.com/waydabber/m1ddc/releases/tag/v1.3.1). It could explain the `110` replies, but the fix was tested on a different monitor, so I can't confirm it resolves this report.

Please retry with the current version and let me know. Leaving this open until confirmed. Thanks!

waydabber, 2026-10-29:
There is still an actionable part here: `m1ddc` decodes read replies without checking the packet format, result code, requested `VCP` code or checksum. Invalid or stale replies can therefore be printed as real values.

#62 fixes the request checksum, but reply validation still needs work. Keeping this open as a `bug` to reject invalid replies and return a useful error. This does not confirm the cause of every hardware report above.

artus70, 2026-10-30:
>     Please retry with the current version and let me know. Leaving this open until confirmed. Thanks!

I updated it via homebrew and `m1ddc display 1 get c `works now!

And also `m1ddc display 1 set input 15` immediately changes the input to dvi as expected - great, thanks!

waydabber, 2026-10-30:
Nice, thanks for the feedback!

closed 2026-10-30 by waydabber