waydabber/m1ddc #41: Display identification based on `EDID UUID` is not reliable

bil0u, 2023-11-21:
Hi there !

I was planning to use your work to automate display input switching alongside [display-switch](https://github.com/haimgel/display-switch) but i faced an issue:

I have two identical displays, same vendor, same model, bought at the same time. It seems that some vendors populate many monitors in batch using the same `EDID UUID`, which leads to some displays not properly recognized (I think issue #35 is related)

https://github.com/waydabber/m1ddc/blob/e22065ef30c923f41fa364af2d3834692750e47c/m1ddc.m#L108

I researched a bit and looked on how Better Display and others are performing display discovery, and while exploring IORegistry is a good option, the only way i could find to have real (different) UUIDs is by doing this

```objective-c
CGDisplayCount getOnlineDisplayInfos(DisplayInfos* displayInfos) {
    CGDisplayCount screenCount;
    CGDirectDisplayID screenList[MAX_DISPLAYS];
    // Retrieving connected displays IDs and count
    CGGetOnlineDisplayList(MAX_DISPLAYS, screenList, &screenCount);

    for (int i = 0; i < (int)screenCount; i++) {
        DisplayInfos *currDisplay = displayInfos + i;
        currDisplay->id = screenList[i];
        // Retrieving display info dictionnary using a private symbol from CoreDisplay Framework
        CFDictionaryRef displayInfos = CoreDisplay_DisplayCreateInfoDictionary(currDisplay->id);
        // Reading actual UUID used by the system, that is persistent across reboots
        currDisplay->uuid = CFDictionaryGetValue(displayInfos, CFSTR("kCGDisplayUUID"));
    }
    return screenCount;
}
```

This is some sample code from the fork i made.

I will open a PR that includes a fix, a code refactor and some extra bonuses !

Happy to discuss about it if needed 🙂

Cheers !

waydabber, 2023-11-21:
Hi there - if the displays are fully identical, the system UUIDs (not talking about the EDID UUID) will not work properly as the OS can change the assignment any time. BetterDisplay by default uses system UUIDs but alternatively it can also read the alphanumerical serial number if present in the EDID (which is not handled by macOS and is sometimes different even when core identifiers are identical) or can also take a look at the hardware port itself and differentiate based on that.

[image]

waydabber, 2023-11-21:
note: for a more advanced matching logic for DDC you can take a look at this - this takes into account the system UUID primarily (but as a fallback can use basic identifiers as well):

https://github.com/waydabber/AppleSiliconDDC/blob/main/AppleSiliconDDC.swift

waydabber, 2023-11-21:
... or not, sorry, this version does not use the system UUID but the framebuffer location (which can be determined by the app which generally uses system UUID to identify a display).

bil0u, 2023-11-21:
Thanks for your prompt answer !

I was initially using the `EDID UUID` alongside the alphanumeric serial number, but i assumed using the system UUID would be more straightforward.

It would be possible to implement all these identification methods pretty easily, as `DisplayInfos` is populated with a lot of these informations, making the selection of methods available with an optional argument.

Currently, i have these informations, but that can be easily extended:
```shell
m1ddc display list detailed
[1] P27q-20 (32F91B73-46A7-4FF5-8E96-1EDBEBF00CE7)
 - Product name:   P27q-20
 - Manufacturer:   LEN
 - AN Serial:      V907WT9W
 - Serial:         0x0000
 - Model:          0x61ea
 - Vendor:         0x30ae
 - Display ID:     3
 - System UUID:    32F91B73-46A7-4FF5-8E96-1EDBEBF00CE7
 - EDID UUID:      30AEEA61-0000-0000-131F-0104A53C2278
 - IO Path:        IOService:/AppleARMPE/arm-io/AppleT600xIO/dispext1@8B000000/AppleCLCD2
 - Adapter:        9475
[2] P27q-20 (3BAF6DBE-5E50-456A-BBB1-F321E8BF8E58)
 - Product name:   P27q-20
 - Manufacturer:   LEN
 - AN Serial:      V907WTAA
 - Serial:         0x0000
 - Model:          0x61ea
 - Vendor:         0x30ae
 - Display ID:     4
 - System UUID:    3BAF6DBE-5E50-456A-BBB1-F321E8BF8E58
 - EDID UUUID:     30AEEA61-0000-0000-131F-0104A53C2278
 - IO Path:        IOService:/AppleARMPE/arm-io/AppleT600xIO/dispext0@88000000/AppleCLCD2
 - Adapter:        9219
```

I'll work on an update !

bil0u, 2023-11-23:
Hi there 👋

PR #42 is updated, the `display` command now supports many identification methods for selecting displays.
An optional prefix can be provided with the identifier, otherwise defaults to `uuid`. Selecting by index still works the same.
```
m1ddc display uuid=32F91B73-46A7-4FF5-8E96-1EDBEBF00CE7 get luminance
m1ddc display edid=30AEEA61-0000-0000-131F-0104A53C2278 get luminance
m1ddc display seid=V907WT9W:30AEEA61-0000-0000-131F-0104A53C2278 get luminance
```

The list of implemented identification methods:
- `id`: `<display_id>`
- `uuid`: `<system_uuid>`
- `edid`: `<edid_uuid>`
- `seid`: `<alphnum_serial>:<edid_uuid>`.
- `basic`: `<vendor>:<model>:<serial>`.
- `ext`: `<vendor>:<model>:<serial>:<manufacturer>:<alphnum_serial>:<product_name>`.
- `full`: `<vendor>:<model>:<serial>:<manufacturer>:<alphnum_serial>:<product_name>:<io_location>`.

Does it make sense for you ?

waydabber, 2023-11-23:
Yes, this sounds great!

waydabber, 2023-11-23:
uhh ohh, I merged the PR but if you'd like, it might be a good idea to add the info to the app help as well (you can open an other PR for that). Thanks for the great contribution!

waydabber, 2023-11-29:
Amazing, thank you!

closed 2023-11-29 by waydabber