waydabber/m1ddc #45: display list got segmentation fault when Air Play to to another Mac

ligolas, 2024-03-03:
[1]    4525 segmentation fault  ./m1ddc display list

ilgoncalves, 2024-08-16:
I’m having the same problem. I usually work with an external monitor and use Continuity with the iPad Pro (formerly Sidecar), which makes the iPad function as an external monitor as well.

[image]
[image]

When I disconnect the MacBook from the iPad, it works normally.

[image]

[image]

@waydabber, maybe this isn’t set up to handle this type of monitor. Could we disable it?

waydabber, 2024-08-16:
Yes, apparently this does not work well indeed. For more advanced CLI based DDC control you can try `betterdisplaycli` - it also has full CLI control to connect/disconnect Sidecar displays as well.

ilgoncalves, 2024-08-16:
> Yes, apparently this does not work well indeed. For more advanced CLI based DDC control you can try `betterdisplaycli` - it also has full CLI control to connect/disconnect Sidecar displays as well.

Thank you for the quick response. I installed the Better Display app, but I couldn’t find `betterdisplaycli` on Homebrew. How can I install it?

waydabber, 2024-08-16:
`brew install waydabber/betterdisplay/betterdisplaycli`

closed 2026-10-29 by waydabber