waydabber/m1ddc #47: What's the current status of supportting the built-in HDMI port of M1 and M2 Macs? Any blockers?

Dentrax, 2024-05-08:
I just tested this tool with M2 Pro Mac using Type-C and I able to switch to HDMI source. Works like a charm. Thanks for such a great tool!

---

As this already pointed out in the [README](https://github.com/waydabber/m1ddc?tab=readme-ov-file#m1ddc) as a _Warning_, I was wondering if there are any updates about HDMI support for M1/M2 Macs:

> _Please note that this tool does not support the built-in HDMI port of M1 and entry level M2 Macs. This tool does not support Intel Macs. You can use [BetterDisplay](https://github.com/waydabber/BetterDisplay#readme) for free DDC control on all Macs and all ports._

I noticed maintainer @waydabber has [dropped a comment](https://github.com/waydabber/m1ddc/issues/27#issuecomment-1492248356) on issue #27:

> _m1ddc does not support the HDMI port of M1 Macs only USB-C. m1ddc is just a very simple app made for testing initially while HDMI support for M1 macs is a bit more complex and not open sourced at this time. BetterDisplay has support as you mentioned and input switching is free._

Questions:
1. Does it support HDMI port with M2 Pro?
2. What do you mean by saying "not open sourced"? Closed source by Apple?
3. Is there any way that we can contribute? How can we add this feature here?
4. I could have download `BetterDisplay` but don't know if it does provide a CLI like what this simple tool does. I'd like to switch display sources using _scripts_, not GUI.
5. Do you know any alternate tools that support this? I found [ddc_hi](https://docs.rs/ddc-hi/latest/ddc_hi/) library and also [ddc-macos-rs](https://github.com/haimgel/ddc-macos-rs) but those are written in Rust.

Thanks in advance.

waydabber, 2024-05-08:
Hi there,

1 - Yes.
2 - No, these are all undocumented stuff
3 - If somebody wishes to implement HDMI support and adds a PR with properly working code, I'll of course accept it.
4 - Yes, BetterDisplay's CLI supports everything m1ddc does + has support for M1/M2 HDMI ports as well as all Intel Macs including the 2018 Mini's HDMI port which is not supported by any other app out there.
5 - Lunar by @alin23 also supports Apple Silicon Macs, including the M1 and M2 HDMI ports and has an easy-to-use CLI (but Lunar must be running, so in this regard it's like BetterDisplay). I wasn't aware of the mentioned libraries - aren't those for Intel only?

waydabber, 2024-05-08:
2. Sure. It's not about whether Swift or Objective-C, can be implemented in both. This feature is just exclusive to BetterDisplay and Lunar for now (as far as I am concerned) and fully available via CLI with both apps if somebody needs it and for most ppl who don't want to use either app the easy fix is is just to use USB-C instead of HDMI (or an USB-C HDMI dongle), so I just put my energies elsewhere. :) But if somebody creates a PR with the required changes, as I said, I'll approve it and will be part of the app, no problem with that.

waydabber, 2026-10-29:
Built-in `HDMI` support for the previously unsupported M1/M2 ports was added in #60 and is now included in [v1.3.1](https://github.com/waydabber/m1ddc/releases/tag/v1.3.1).

Closing this as implemented. Thanks!

closed 2026-10-29 by waydabber