waydabber/m1ddc #49: m1ddc get input always returns the same value

erentantekin, 2024-12-01:
I'm trying to get the current input value but it seems to always return 15.

```
m1ddc get input
15
```

I'm able to successfully switch inputs between HDMI and USB-c using set input-alt. But I can't read the current input.

Macbook Pro M3 Max
MacOS 14.7.1
Latest version of m1ddc from homebrew

waydabber, 2026-10-29:
A bug that could make `get` repeat an old reply was fixed in #62 and is included in [v1.3.1](https://github.com/waydabber/m1ddc/releases/tag/v1.3.1). It was tested on a different monitor, so please retry `get input` with the current version.

Also, `get input-alt` is not reliable on all LG displays, as noted in #32. Leaving this open until we know whether the fix resolves your case. Thanks!

waydabber, 2026-10-29:
The request checksum bug that could cause stale reads was fixed in #62 and is included in `v1.3.1`. Some LG displays also do not provide useful `get input-alt` replies, as noted in #32.

The remaining software work to validate replies and reject stale or invalid data is now tracked in #37, so closing this here to keep that work in one place. A valid reply that reports the wrong input may still be a monitor firmware limitation. Thanks!

closed 2026-10-29 by waydabber