waydabber/m1ddc #50: LG monitor: can't set input?

zapman449, 2025-02-03:
mac M3 laptop, windows desktop, LG 38wn95c-w monitor, attempting to switch input back and forth from the mac.  Mac connected by usb-c provided cable, windows desktop on Display Port

```
❯ ./m1ddc display list detailed
...
[2] LG HDR WQHD+ (2CA2 .. uuid .. BB3)
 - Product name:  LG HDR WQHD+
 - Manufacturer:  GSM
 - AN Serial:     406NTFA...
 - Vendor:        7789 (0x1e6d)
 - Model:         30542 (0x774e)
 - Serial:        166291 (0x28993)
 - Display ID:    2
 - System UUID:   2CA23 .. uuid .. BB3
 - EDID UUID:     1E6D .. uuid .. 578
 - IO Location:   IOService:/AppleARMPE/arm-io@10F00000/AppleH15IO/dispext0@4000000/IOMobileFramebufferShim
 - Adapter:       16131
```

I can get the screen to flash when I run `./m1ddc display 2 set input-alt $x` where x is 144, 145, 208, 209 and 210, but the input isn't set to the windows box... just comes back to the mac.

Any suggestions?

waydabber, 2026-10-29:
The LG `input-alt` write checksum was fixed in #52 and is included in [v1.3.1](https://github.com/waydabber/m1ddc/releases/tag/v1.3.1). Please try `m1ddc display 2 set input-alt 208` again with the current version for the `DisplayPort` input.

I don't have confirmation for the `38WN95C-W`, so leaving this open until we know whether that fixes it. Thanks!

waydabber, 2026-10-29:
The known LG `input-alt` checksum issue is fixed in #52 and included in `v1.3.1`, as noted above. Beyond that, accepted input values and switching behavior depend on the monitor firmware and whether the destination input has a usable signal.

Closing this as a model-specific support case, since we do not have a reproduction of a remaining code bug. If the current version still fails while another tool works on the same connection, please open a new report with the exact working and failing commands. Thanks!

closed 2026-10-29 by waydabber